CommunityGD32Cores/platform-gd32 #17: Any way to support Sipeed-RV-Debugger for debug and firmware upload

richardclli, 2021-10-02:
This debugger can be used in GD32VF series, not sure if it can work on GD32F series.  I can help to work out the development, but need some advice.

maxgerhardt, 2021-10-02:
Can you connect to the chip with that adapter and some OpenOCD invocation on the commandline? 

Please try:
* Go into the folder `<home path>/.platformio/packages/tool-openocd/scripts/interface/ftdi`, create a new file `sipeed-rv-debugger.cfg` there with the content 
```
interface ftdi
ftdi_vid_pid 0x0403 0x6010
ftdi_channel 0
ftdi_layout_init 0x00e8 0x00eb

transport select swd
ftdi_tdo_sample_edge falling
```

What does

```
C:\Users\<user>\.platformio\packages\tool-openocd\bin\openocd -f interface/ftdi/sipeed-rv-debugger.cfg -c "set CPUTAPID 0" -f target/stm32f1x.cfg`
```

output? (Change path accordingly, debugger connected to the target chip via SWDIO+SWDCLK+GND+3.3V powered, maybe NRST connected too)

You may also try via JTAG instead of SWD, change to `transport select jtag` in the `.cfg` file and retry.

richardclli, 2021-10-02:
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.11.0-00155-ge392e485e (2021-03-15-16:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
0
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : JTAG tap: stm32f1x.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x4)
Info : JTAG tap: stm32f1x.bs tap/device found: 0x790007a3 (mfg: 0x3d1 (GigaDevice Semiconductor (Beijing) Inc), part: 0x9000, ver: 0x7)
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Error: stm32f1x.cpu -- clearing lockup after double fault
Polling target stm32f1x.cpu failed, trying to reexamine
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections

richardclli, 2021-10-02:
The config file that I used is:

interface ftdi
ftdi_device_desc "JTAG Debugger"
ftdi_vid_pid 0x0403 0x6010

#autoexit true

#interface cmsis-dap
transport select jtag
ftdi_layout_init 0x0008 0x001b
ftdi_layout_signal nSRST -oe 0x0020 -data 0x0020

richardclli, 2021-10-02:
modified from GD32VF103 settings.

maxgerhardt, 2021-10-02:
Okay that looks good. Which exact target chip are you using? We can set-up a quick test.

richardclli, 2021-10-02:
GD32F103TBU6

maxgerhardt, 2021-10-02:
Can you do the following:

1. Open a [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and execute `pio platform update gd32`
2. Make sure to delete (or move) your old `C:\Users\<user>\.platformio\packages\tool-openocd\scripts\interface\ftdi\sipeed-rv-debugger.cfg` file
3. Clone and open the https://github.com/CommunityGD32Cores/gd32-pio-projects/tree/main/gd32-spl-blinky project in VSCode + PlatformIO
4. Change the complete `platformio.ini` file to 
```ini
[env]
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
platform_packages = 
    framework-spl-gd32@https://github.com/CommunityGD32Cores/gd32-pio-spl-package.git

[env:genericGD32F103TB]
board = genericGD32F103TB
framework = spl
upload_protocol = sipeed-rv-debugger
debug_tool = sipeed-rv-debugger
```
5. Use the [project environment switcher](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks) to select the right project and the `genericGD32F103TB` environment
6. Execute the "Upload" project task. What's the output?
7. If that was successfull, try debugging. (Left "Debugging" sidebar, make sure "PIO Debug (gd32-spl-blinky)" is the selected configuration, watch the "Debug Console" output for potential error messages

richardclli, 2021-10-02:
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.11.0-00155-ge392e485e (2021-03-15-16:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

0
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 00000000 msp: 00000000
** Programming Started **
Error: JTAG-DP STICKY ERROR
Error: Failed to read memory at 0x1ffff7e2
Warn : STM32 flash size failed, probe inaccurate - assuming 128k flash
Error: stm32x device protected
Error: failed erasing sectors 0 to 6
embedded:startup.tcl:530: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 595
at file "embedded:startup.tcl", line 530
*** [upload] Error 1

richardclli, 2021-10-02:
Not sure it is a hardware problem or software problem.

maxgerhardt, 2021-10-02:
The flash is hardware write protected, you need to execute a OpenOCD command to unlock it once. Let me search for the command.

maxgerhardt, 2021-10-02:
Execute this after adapting the path to OpenOCD.
```
C:\Users\<user>\.platformio\packages\tool-openocd\bin\openocd -f interface/ftdi/sipeed-rv-debugger.cfg -c "set CPUTAPID 0" -f target/stm32f1x.cfg -c "init; reset halt; stm32f1x unlock 0; reset halt; exit"
```

then try upload again. We will add a shortcut button for "Unlock Flash" in the future.

richardclli, 2021-10-05:
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.11.0-00155-ge392e485e (2021-03-15-16:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

0
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: stm32f1x.cpu: IR capture error; saw 0x0f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Invalid ACK (7) in DAP response
Error: JTAG-DP STICKY ERROR
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1

maxgerhardt, 2021-10-05:
No now something is wrong with the JTAG wiring or power, it shouldn't say 

>Error: JTAG scan chain interrogation failed: all ones

Try executing double checking all wiring, or playing around with reset (e.g., manually connect it to GND before OpenOCD starts up, then release it when OpenOCD starts to connect) while executing the first command that just connects to the chip regularly. As long as you don't again see 

>Info : JTAG tap: stm32f1x.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x4)
>Info : JTAG tap: stm32f1x.bs tap/device found: 0x790007a3 (mfg: 0x3d1 (GigaDevice Semiconductor (Beijing) Inc), part: 0x9000, ver: 0x7)

Something is wrong.

maxgerhardt, 2021-10-10:
Great, I'll see to it that the board definition autogenerator enables that adapter for all boards.

maxgerhardt, 2021-10-15:
Resolved in 0e20c2bf8c9a1e06a9ad7ad61b5f560ec6909cb8.

closed 2021-10-15 by maxgerhardt