CommunityGD32Cores/platform-gd32 #18: Uploading / Debugging GD32E23x fails with configurations using OpenOCD

djix123, 2021-11-13:
OpenOCD currently does not support the GD32E23x line of processors (there are patches pending, but not yet accepted - I've a minimally working patch set if there's interest).

On the other hand, Blackmagic Probe (recent builds) do work for these boards. I would propose that 'blackmagic' is the default upload / debug tool for the GD32E23x until OpenOCD adds support.

I can submit a patch/pull request to change the default tool if helpful.

maxgerhardt, 2021-11-14:
1. Have you verified within a PlatformIO project that `upload_protocol = blackmagic` in the `platformio.ini` makes uploading work with a black-magic probe? 
2. Does debugging work too if you add `debug_tool = blackmagic`? 

>(there are patches pending, but not yet accepted - I've a minimally working patch set if there's interest).
3. The official GigaDevice OpenOCD fork has in fact GD32E23x support (see https://github.com/GigaDevice-Semiconductor/openocd/issues/5), did you create your own patches or did you use that?

djix123, 2021-11-14:
Yes, I've tested with `upload_protocol = blackmagic` and `debug_tool = blackmagic` both uploading and debugging work as expected.

Uploading:

Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.6% (used 36 bytes from 6144 bytes)
Flash: [          ]   3.3% (used 1092 bytes from 32768 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, serial, sipeed-rv-debugger, stlink
CURRENT: upload_protocol = blackmagic
Looking for BlackMagic port...
Use manually specified: /dev/cu.usbmodem14501
Uploading .pio/build/genericGD32E230F6/firmware.elf
Target voltage: 3.28V
Available Targets:
No. Att Driver
 1      GD32E230 M23
delay_1ms (count=count@entry=50) at src/main.c:52
52          }
Loading section .isr_vector, size 0xcc lma 0x8000000
Loading section .text, size 0x440 lma 0x80000cc
Loading section .init_array, size 0x4 lma 0x800050c
Loading section .fini_array, size 0x4 lma 0x8000510
Loading section .data, size 0x4 lma 0x8000514
Start address 0x800011c, load size 1304
Transfer rate: 17 KB/sec, 217 bytes/write.
Section .isr_vector, range 0x8000000 -- 0x80000cc: matched.
Section .text, range 0x80000cc -- 0x800050c: matched.
Section .init_array, range 0x800050c -- 0x8000510: matched.
Section .fini_array, range 0x8000510 -- 0x8000514: matched.
Section .data, range 0x8000514 -- 0x8000518: matched.
Kill the program being debugged? (y or n) [answered Y; input not from terminal]
[Inferior 1 (Remote target) killed]

Debugging:

Reading symbols from /Users/jgiles/SynologyDrive/Documents/MCU/projects/GD32E230/.pio/build/genericGD32E230F6/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = blackmagic
PlatformIO: Initializing remote target...
Target voltage: 3.28V
Available Targets:
No. Att Driver
 1      GD32E230 M23
0x080001c8 in delay_1ms (count=count@entry=500) at src/main.c:52
52	    }
Loading section .isr_vector, size 0xcc lma 0x8000000
Loading section .text, size 0x440 lma 0x80000cc
Loading section .init_array, size 0x4 lma 0x800050c
Loading section .fini_array, size 0x4 lma 0x8000510
Loading section .data, size 0x4 lma 0x8000514
Start address 0x800011c, load size 1304
Transfer rate: 17 KB/sec, 217 bytes/write.
Temporary breakpoint 1 at 0x80001d0: file src/main.c, line 13.
Warning: the current language does not match this frame.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> https://bit.ly/pio-debug
Note: automatically using hardware breakpoints for read-only addresses.

Temporary breakpoint 1, main () at src/main.c:13
13	{

djix123, 2021-11-14:
The GigaDevice OpenOCD fork is no longer current with latest OpenOCD.

The minimal patch set I create were based on:

Additions to flash/nor/stm32f1x.c:
https://review.openocd.org/c/openocd/+/5246/ 

GD32E23x.cfg:
https://review.openocd.org/c/openocd/+/6543

Needed to rework '5246' as its a patch against an older version of stm32f1x.c than current, and revert GD32E23x.cfg from '6543' to use 'stm32f1' as the flash driver since that patch creates a new gd32 driver.

I really wanted something that works versus HEAD openocd so I don't have to install multiple versions and then remember which to use and when.

maxgerhardt, 2021-11-14:
Thanks for all the input, I'll have a play with trying to get the GigaDevice OpenOCD changes for GD32E23x (which look like literally [2 clean commits](https://github.com/GigaDevice-Semiconductor/openocd/commits/xpack) into the latest OpenOCD, I'm sure the GigaDevice people would also like that so that they're updated. 

>I really wanted something that works versus HEAD openocd so I don't have to install multiple versions and then remember which to use and when.

For PlatformIO that's thankfully not a problem because of internal packet management -- PIO never uses the system's `openocd` version, we can point it at any pre-prepared binary distribution of OpenOCD we want to, e.g., the one with the GD32 patches. Currently we use the standard-provided OpenOCD and remap GD32 targets to STM32, which is however only possible in some cases (not working for E23x and others).

If I can compile such an OpenOCD version (https://github.com/maxgerhardt/openocd/pull/1), would you be available to test it on real HW with an ST-Link (or "GDLink")? I only have a GD32E1xx board I think. If that works we have more complete OpenOCD support, your BMP configuration is unaffected by that. 

If I fail to do that, we'll switch the default for the boards that have no working OpenOCD config to BMP.

djix123, 2021-11-14:
Yes, if you can compile an updated OpenOCD version with the GD32 patches will happily test. I don't have a GDLink, but do have an ST-Link and a generic CMSIS-DAP device as well.

djix123, 2021-11-17:
Just notice that the OpenOCD project accepted a patch to support the GD32E230 today. Looks difference from GigaDevices patch, though.

maxgerhardt, 2021-12-25:
We've switched the OpenOCD version to the one provided by the latest xPack build with GD32E23x support in fbaebc9317c24e77c2ae06c3365d8aba122f639a, closing as solved.

closed 2021-12-25 by maxgerhardt