CommunityGD32Cores/platform-gd32 #21: GD32L23x support

maxgerhardt, 2021-12-19:
The new L23x low-power Cortex-M23 series of GigaDevice microcontrollers (https://www.gigadevice.com/press-release/gigadevice-launches-the-new-gd32l233-series-for-energy-saving-applications/) should be supported in this platform by providing the necessary board definitions for all chips of that series, SVD, upload methods, builder logic et cetera.

Right now, preliminary, untested support is available for the GD32L233C-START board via https://github.com/CommunityGD32Cores/platform-gd32/blob/main/boards/gd32l233c_start.json and the associated example projects.

plakkiesza, 2022-04-19:
I have a GD32L233K-START board and this helped me get it working. I just had to change the following.
Created a new board file based on gd32l233c_start.json(this was in .platformio/platforms/dg32/boards). I just changed the max memory to the correct value and a few i think not related stuff so it fits as a new file. I called this file gd32l233k_start.json
`{
  "build": {
    "core": "gd32",
    "cpu": "cortex-m23",
    "extra_flags": "-DGD32L2 -DGD32L233 -DGD32L23x",
    "f_cpu": "64000000L",
    "mcu": "gd32l233kbt6",
    "spl_series": "GD32L23x",
    "series": "GD32L233"
  },
  "debug": {
    "default_tools": [
      "cmsis-dap"
    ],
    "jlink_device": "GD32L233KB",
    "onboard_tools": [
      "cmsis-dap"
    ],
    "openocd_target": "gd32e23x",
    "svd_path": "GD32L23x.svd",
    "openocd_extra_pre_target_args": [
      "-c",
      "set CPUTAPID 0"
    ]
  },
  "frameworks": [
    "spl"
  ],
  "name": "GD32L233CC (24k RAM, 128k Flash)",
  "upload": {
    "disable_flushing": false,
    "maximum_ram_size": 24576,
    "maximum_size": 131072,
    "protocol": "cmsis-dap",
    "protocols": [
      "jlink",
      "cmsis-dap",
      "stlink",
      "blackmagic",
      "sipeed-rv-debugger",
      "serial"
    ],
    "require_upload_port": true,
    "use_1200bps_touch": false,
    "wait_for_upload_port": false
  },
  "url": "https://www.gigadevice.com/press-release/gigadevice-launches-the-new-gd32l233-series-for-energy-saving-applications/",
  "vendor": "GigaDevice"
}`
This is only for the spl_blinky sketch.
I also added the new board to the platform.ini file. 

> [env:gd32l233k_start]
>board = gd32l233k_start
>framework = spl

I added the following ifdef to the main.c file.

>/* define blinky LED pin here, board specific, otherwise default PC13 */
>#ifdef GD32350G_START
>/* correct LED for GD32350G-START board. PA1 */
>#define LEDPORT     GPIOA
>#define LEDPIN      GPIO_PIN_1
>#define LED_CLOCK   RCU_GPIOA
>#elif GD32L233K_START
>/* correct LED for GD32L233k-START board. PA8 */
>#define LEDPORT     GPIOA 
>#define LEDPIN      GPIO_PIN_8 
>#define LED_CLOCK   RCU_GPIOA 
>#else
>#define LEDPORT     GPIOC
>#define LEDPIN      GPIO_PIN_13
>#define LED_CLOCK   RCU_GPIOC
>#endif

maxgerhardt, 2022-04-19:
I've added this board JSON to our repo and added it in the examples to CI in d38af82b96ab34b02d17eadbe380f5059969a267, 365219ce412722570b50b052f803740d6dff17b2, 81dec41c615fa0de7610b8bd3a6bbf39cf02c001, let's see if this buids and works.

Did you test live debugging? (Left debugging sidebar -> Play button). We currently treat this L23x board like E23x, but since you were able to upload it might work (?)

plakkiesza, 2022-04-19:
I have, and it works perfectly as far as I can tell. Very cool. The new board file did the trick.

maxgerhardt, 2022-04-19:
Great! I'll leave this issue open until we have all board JSON files for the chips + official dev boards.

Thankfully https://www.gigadevice.com/products/microcontrollers/gd32/arm-cortex-m23/?fwp_processor_type=cortex-m23&fwp_microcontrollers_product_series=gd32l233 was now expanded (which I was waiting for) so our auto-generator can be fed with the new CSVs.

maxgerhardt, 2022-04-19:
Added missing boars in d9b9c987b32cb0ea4c6ecc74ae7210e69c3883d0 and 62cdfa5f93ef59af120d4dceda9a29153627cab7, closing.

closed 2022-04-19 by maxgerhardt