CommunityGD32Cores/platform-gd32 #22: Does CMSIS-DSP integration work already?

crosswick, 2021-12-19:
I'm looking to use some functions from CMSIS-DSP; is this already possible?

I couldn't find Gigadevice chips in https://docs.platformio.org/en/latest/frameworks/cmsis.html#platforms ...

maxgerhardt, 2021-12-20:
We're not an official platform made by PlatformIO so we're not showing up in their docs yet before we request it. 

I think for the SPL and Arduino framework, we only include the header for CMSIS-DSP / arm_math.h but don't link in the library code, so compilation will fail. It's however easy to add this in.

What target microcontroller and framework (SPL, Arduino, mbedos) are you working on?

crosswick, 2021-12-20:
I see. GD32E103CBT6 and GD32F150C8T6 for now, both SPL.

maxgerhardt, 2021-12-20:
Very well, I'll prioritise those -- should have support for that by the end of tomorrow.

crosswick, 2021-12-20:
:) awesome, thank you.

maxgerhardt, 2021-12-21:
I've added the CMSIS-DSP library version 5.7.0 (last one to directly have the precompiled library files in the download...) in our SPL package (https://github.com/CommunityGD32Cores/gd32-pio-spl-package/commit/4e586eae3656ee0c49210f187b7f3fc057c16820) and expanded the SPL builder script to auto-discover this library and set the correct FPU and float-ABI flags (https://github.com/CommunityGD32Cores/platform-gd32/commit/2707955a346211033f67015f54faadb3677d3124). 

I've created an example project https://github.com/CommunityGD32Cores/gd32-pio-projects/tree/main/gd32-spl-cmsis-dsp inspired by the official example https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP/Examples/ARM/arm_sin_cos_example and tested it on a GD32F350G8 board, where it works nicely. Compilation for your two chips was also tested, but I don't have the hardware for them at hand. 

Please use [the CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) to execute `pio platform update gd32` and import and test the linked `gd32-spl-cmsis-dsp` project regarding the `genericGD32E103CB` and `genericGD32F150C8` environments (use the [project environment switcher](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks) to switch). Does it work?

crosswick, 2021-12-21:
Great!!! I will test it asap. Thank you very much

crosswick, 2021-12-23:
These two boards work! GD32F150C8 and GD32E103C8, great - haven't been able to test my own GD32E103CB-based design yet, but I think that shouldn't be a problem. 

The GD32E230C4 config doesn't compile, but I guess Cortex-M23 isn't supported yet? I saw this one commented out in the platformio.ini, I have a GD32E230C8 board that I could test with. 

Thank you!

maxgerhardt, 2021-12-23:
The E23x series with its Cortex-M23 is supported in the general build system, but the specific GD32E230C4 chip only has 16 Kilobytes of flash, which is not enough for the SPL + CMSIS-DSP + printf (with floating pionter support) demo. 

>ld.exe: region `FLASH' overflowed by 12112 bytes

by about 12 kilobytes. 

If you take out the two `printf()` calls and change the `platformio.ini` option to `board_build.spl_printf_float = no`, then the firmware fits, but has no `printf()` output (which might or might not matter).

```
RAM:   [          ]   1.6% (used 64 bytes from 4096 bytes)
Flash: [=====     ]  49.9% (used 8180 bytes from 16384 bytes)
```

(this also shows how excessively big the printf() implementation, even for the newlib-nano configuration, is).

Thus it's not a fault of the platform really that compilation for that chip fails. The next bigger C6 chip (32 kBytes) works out-of-the box with all intended features, but its flash is also close to saturated.

```
RAM:   [=         ]   9.0% (used 556 bytes from 6144 bytes)
Flash: [========= ]  86.3% (used 28284 bytes from 32768 bytes)
```

crosswick, 2021-12-23:
Alright, so it should be relatively easy for me to get my GD32E230C8 board to work with this example? I wouldn't immediately know how myself.

maxgerhardt, 2021-12-23:
Exactly. As per [available boards](https://github.com/CommunityGD32Cores/platform-gd32/tree/main/boards) you can add to the `platformio.ini` just the environment for your board

```ini
[env:genericGD32E230C8]
board = genericGD32E230C8
framework = spl
```

and use the project environment switcher again as linked above to compile and upload for this new environment. 

Compilation should result in

``` 
RAM:   [=         ]   6.8% (used 556 bytes from 8192 bytes)
Flash: [====      ]  43.2% (used 28284 bytes from 65536 bytes)
Building .pio\build\genericGD32E230C8\firmware.bin
========= [SUCCESS] Took 2.61 seconds =========
```

crosswick, 2021-12-23:
False - it took 4.78 seconds :P

maxgerhardt, 2021-12-23:
Okay well it seems you have circa half the CPU cores or processing power as I have, but the general successful compilation result is the same. 

I'll close this issue as solved through https://github.com/CommunityGD32Cores/platform-gd32/commit/2707955a346211033f67015f54faadb3677d3124 then. 

Other CMSIS libraries like CMSIS-NN will likely be added too soon. Maybe also also a "compile from source" option for CMSIS-DSP instead of the current precompiled way.

closed 2021-12-23 by maxgerhardt