CommunityGD32Cores/platform-gd32 #25: Wrong openocd_target for GD32F205

PetteriAimonen, 2022-01-04:
GD32F205 seems to need `openocd_target = stm32f1x` instead of `openocd_target = stm32f2x` here: https://github.com/CommunityGD32Cores/platform-gd32/blob/main/boards/genericGD32F205VC.json

With the current setting `stm32f2x` the following error occurs when running `pio run -v --target upload`:

```
Info : device id = 0x15080418
Warn : Cannot identify target as a STM32 family.
```

Even though the `CPUTAPID=0` override ignores the CPU ID mismatch, openocd seems to be unable to identify the correct flashing algorithm. But when `stm32f1x` is specified, it works both with and without the `CPUTAPID` override.

maxgerhardt, 2022-01-04:
Fixed in https://github.com/CommunityGD32Cores/platform-gd32/commit/7b247d658d2dfe926f4cf59d27f612c2418324eb.

closed 2022-01-04 by maxgerhardt

PetteriAimonen, 2022-01-04:
Thanks for the quick fix, works great now!