CommunityGD32Cores/platform-gd32 #33: Suport for GD32VF (RISCV) series.

richardclli, 2022-08-19:
I can contribute, need somebody give me some ideas where to start.

I already PlatformIO projects works for GD32VF series.  Just want to unified to a single opensource and active platform.

maxgerhardt, 2022-08-19:
I'm not sure about this -- wouldn't we be doing duplicate work that https://github.com/sipeed/platform-gd32v has already done? What are is that platform missing, feature wise? I see the RISC chips as very different from the ARM chips and we would basically have different basic chip types in one platform, which I've not yet seen in PlatformIO.

richardclli, 2022-08-22:
The main problem is the sdk for GD32VF103 seems not maintained anymore: https://github.com/riscv-mcu/GD32VF103_Firmware_Library , bug fixes pull request is not processed anyway.

And for compatibility, the code will run in many cases for chips GD32VF103 / GD32E103 / GD32F103 / GD32C103, as the firmware library has 99% compatible API.

maxgerhardt, 2022-08-22:
And when we have SPL support, won't people also be asking for Arduino support through https://github.com/sipeed/Longduino? Or should we adapt https://github.com/CommunityGD32Cores/ArduinoCore-GD32 to also support the RISC-V chips?

djix123, 2022-08-22:
I think its a little more tricky than saying the SPL is 99% compatible. The compiler chain and compiler command line options are different and so is the start up assembler file. Effectively it would require install two different tool-chains under one platform and then having to differentiate between them in the build scripts. Not sure that's a good idea.

Also the SPL is maintained: according to this site https://www.gd32mcu.com/en/download/7?kw=GD32VF1 it was last updated July 12, 2022.

maxgerhardt, 2022-08-22:
The SPL is indeed maintened from the Gigadevice side, but the sipeed paltform references

https://github.com/sipeed/platform-gd32v/blob/master/platform.json#L43-L48

and that was last update a whopping 3 years ago (https://registry.platformio.org/tools/platformio/framework-gd32vf103-sdk/versions), so richard has a point there.

But we could still do a PR into the sipeed platform, which Ivan seems to maintain according to the last commit.

But then there still wouldn't be general support for all possible GD32VF103 chip variants.

djix123, 2022-08-22:
I did actually start doing that: https://github.com/djix123/framework-gd32vf103-sdk but didn't get around to submitting a PR!

richardclli, 2022-08-23:
I have worked out something as well, but.... I think it is better to do it in https://github.com/CommunityGD32Cores for completeness of support for all GD32 chips.

And for the platform, it will be no harm to separate 2 platforms for different tool chains, and after taken care the underlying layer, the upper layer (e.g. Arduino) maybe able to support both platforms easily.