CommunityGD32Cores/platform-gd32 #42: Upload fails in Platformio and ST-Util

IerwinDE, 2023-03-09:
Hello,

I'm trying to flash my Sideboard that is equipped with a GD32F130-C8T6.
After I realized that the programmig pins are in a different order than in the description I have successfully unlocked the chip.
There was an error saying Option Bytes could not be set but since the same error is shown in the video that is linked in the redme I guess thats ok.
When I try to upload with PlatformIO I am getting an error after the compilation.
I have very little knowledge of PlatformIO or Visual Studio but the message seems to suggest that its not finding the conected device and insteag looks for a remote agent to perform the flashing:
```
Building project locally
Processing VARIANT_HOVERCAR (board: genericGD32F130C6; framework: spl; platform: https://github.com/CommunityGD32Cores/platform-gd32.git#e712045bdba54790ed2e07c47a421345340efb31)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/gd32/genericGD32F130C6.html
PLATFORM: GD GD32 (1.0.0+sha.e712045) > Generic GD32F130C6T6
HARDWARE: GD32F130C6T6 72MHz, 4KB RAM, 32KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES:
 - framework-spl-gd32 @ 2.10000.0+sha.67a6fe9
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
SPL libraries are included: True
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\VARIANT_HOVERCAR\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.5% (used 636 bytes from 4096 bytes)   
Flash: [=======   ]  70.3% (used 23040 bytes from 32768 bytes)
========================================================================================= [SUCCESS] Took 1.84 seconds =========================================================================================

Environment       Status    Duration
----------------  --------  ------------
VARIANT_HOVERCAR  SUCCESS   00:00:01.837
========================================================================================= 1 succeeded in 00:00:01.837 =========================================================================================
Uploading firmware remotely
Could not find active agents. Please start it before on a remote machine using `pio remote agent start` command.
See http://docs.platformio.org/page/plus/pio-remote.html
```

I also tried to flash the firmware.bin from platformio via ST-Link and get an error message:
```
08:32:41 : ST-LINK SN : 280044001000004A4735544E
08:32:41 : V2J37S7
08:32:41 : Connected via SWD.
08:32:41 : SWD Frequency = 4,0 MHz.
08:32:41 : Connection mode : Normal.
08:32:41 : Debug in Low Power mode enabled.
08:32:41 : Device ID:0x410 
08:32:41 : Device flash Size : 64KBytes
08:32:41 : Device family :STM32F10xx Medium-density
08:32:46 : Unexpected error
08:32:46 : Error occured during program operation!
08:32:46 : Programming error @ 0x08000000!
08:32:48 : Programmed memory Checksum: 0x001DE200
```

IerwinDE, 2023-03-09:
for some reaosn PlatformIO only showed me the Link for Remote Update. Once I clicked "show all tasks" there was the regular non remote update Task and that worked like a charm

maxgerhardt, 2023-03-09:
>for some reaosn PlatformIO only showed me the Link for Remote Update

That's weird. Can you restart VSCode again and show a screenshot of the available project tasks?

closed 2025-08-15 by maxgerhardt