CommunityGD32Cores/platform-gd32 #44: framework = arduino Not work , framework = spl OK

QGB, 2023-04-18:
even simple gpio
```
#define LED PB9

void setup(){
    pinMode(LED, OUTPUT);
    digitalWrite(LED, LOW);
    delay(500);
    digitalWrite(LED, HIGH);
    delay(500);
    Serial1.begin(57600);
}
```
spl code work well
```
#include "gd32f1x0.h"

#define LEDPORT     GPIOB
#define LEDPIN      GPIO_PIN_9
#define LED_CLOCK   RCU_GPIOB

int main(void){
    systick_config();
    rcu_periph_clock_enable(LED_CLOCK);

    gpio_mode_set(LEDPORT, GPIO_MODE_OUTPUT, GPIO_PUPD_NONE, LEDPIN);
    gpio_output_options_set(LEDPORT, GPIO_OTYPE_PP, GPIO_OSPEED_2MHZ, LEDPIN);

    while (1){
        gpio_bit_reset(LEDPORT, LEDPIN);//buzzer
        delay_1ms(100);

        gpio_bit_set(LEDPORT, LEDPIN);
        delay_1ms(500);
    }
}
```

maxgerhardt, 2023-04-18:
Well technically for Arduino blinky to have the same behavior you would have
```cpp
#define LED PB9

void setup(){
    pinMode(LED, OUTPUT);
}

void loop() {
    digitalWrite(LED, LOW);
    delay(500);
    digitalWrite(LED, HIGH);
    delay(500);
}
```

To reproduce this, I would also need to know which exact board you're compiling for. Please post the exact `platformio.ini`.

QGB, 2023-04-18:
arduino
```
[env]
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
platform_packages = 
    framework-arduinogd32@https://github.com/CommunityGD32Cores/ArduinoCore-GD32.git
monitor_speed = 115200

[env:genericGD32F130C6]
board = genericGD32F130C6
framework = arduino
```

spl
```
[env]
platform = https://github.com/CommunityGD32Cores/platform-gd32.git
platform_packages = 
    framework-spl-gd32@https://github.com/CommunityGD32Cores/gd32-pio-spl-package.git

[env:genericGD32F130C6]
board = genericGD32F130C6
framework = spl
```

maxgerhardt, 2023-04-18:
What exact board are you running on? Custom made? Does it have a quartz crystal for HSE?

QGB, 2023-04-18:
> What exact board are you running on? Custom made? Does it have a quartz crystal for HSE?

NO external quartz crystal

maxgerhardt, 2023-04-18:
Hm okay that should still be fine because the Arduino  system code configures the clock to be sourced from the internal RC 8MHz oscillator.

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/41f3c2b3524a51e698ef03186a4243bb89e8ad88/system/GD32F1x0_firmware/CMSIS/GD/GD32F1x0/Source/system_gd32f1x0.c#L45-L49

Can you test this exact code? https://github.com/CommunityGD32Cores/platform-gd32/issues/44#issuecomment-1512875020

If you're uploading via ST-Link, you should also be able to use the Debugging sidebar ("PIO Debug"). Does it get stuck somewhere?

QGB, 2023-04-18:
Your code works,  How to config Serial1 pin PA2 as RX, PA3 as TX

maxgerhardt, 2023-04-18:
This should already be the default configuration for `Serial1` as set by

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/41f3c2b3524a51e698ef03186a4243bb89e8ad88/variants/GD32F130C6_GENERIC/variant.h#L130-L137

And the pin mapping looks good too

https://github.com/CommunityGD32Cores/ArduinoCore-GD32/blob/41f3c2b3524a51e698ef03186a4243bb89e8ad88/variants/GD32F130C6_GENERIC/PeripheralPins.c#L180-L198

When you go back to your original code and debug it in the debugger, do you see it getting stuck somewhere?

QGB, 2023-04-18:
```
#include <Arduino.h>

#define LED PB9

void setup(){
    pinMode(LED, OUTPUT);
    Serial1.begin(57600);
    Serial1.println("Start on Serial1!");

}

void loop() {
    digitalWrite(LED, LOW);
    delay(500);
    digitalWrite(LED, HIGH);
    delay(50);
    Serial1.println("Serial1 \r\n");
}
```

gpio OK, but no output in serial console . I try switch RX TX cable.also not work

maxgerhardt, 2023-05-01:
I checkked the code again and the macros are named very confusingly indeed. 

[image]

So you actually need to use `Serial2`.

```cpp
#include <Arduino.h>

#define LED PB9

void setup(){
    pinMode(LED, OUTPUT);
    Serial2.begin(57600);
    Serial2.println("Start on Serial2!");

}

void loop() {
    digitalWrite(LED, LOW);
    delay(500);
    digitalWrite(LED, HIGH);
    delay(50);
    Serial2.println("Serial2");
}
```

closed 2025-08-15 by maxgerhardt