CommunityGD32Cores/platform-gd32 #45: Uploading and Debugging gor e505 fails with OpenOCD

MarvFPV, 2023-04-18:
Hiya, 
am getting an issue where i can't seem to get opencd to work on a gd32e505re.

This is the error message:
```
Building in release mode
Checking size .pio\build\genericGD32E505RE\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.0% (used 36 bytes from 131072 bytes)
Flash: [          ]   0.2% (used 976 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, gdlinkcli, jlink, serial, sipeed-rv-debugger, stlink
CURRENT: upload_protocol = cmsis-dap
Uploading .pio\build\genericGD32E505RE\firmware.elf
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-12-07-17:33)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

swd
0
embedded:startup.tcl:26: Error: Can't find target/gd32e50x.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 26
*** [upload] Error 1
```

I tried manually adding the gd32e50x.cfg from [here](https://review.openocd.org/c/openocd/+/6795/1/tcl/target/gd32e50x.cfg#52https://review.openocd.org/c/openocd/+/6795/1/tcl/target/gd32e50x.cfg#52) but that gives me this error message:
```
Building in release mode
Checking size .pio\build\genericGD32E505RE\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.0% (used 36 bytes from 131072 bytes)
Flash: [          ]   0.2% (used 976 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, gdlinkcli, jlink, serial, sipeed-rv-debugger, stlink
CURRENT: upload_protocol = cmsis-dap
Uploading .pio\build\genericGD32E505RE\firmware.elf
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-12-07-17:33)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

swd
0
target halted due to debug-request, current mode: Thread 
xPSR: 0xf9000000 pc: 0x080001e0 msp: 0x20020000
** Programming Started **
Error: Cannot identify target as a stm32x
Error: auto_probe failed
embedded:startup.tcl:1136: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 1201
at file "embedded:startup.tcl", line 1136
*** [upload] Error 1
```
For reference, this is what my env looks like:
```
[env:genericGD32E505RE]
board = genericGD32E505RE
framework = spl
upload_protocol = cmsis-dap
debug_tool = cmsis-dap
```

Any input on what i'm doing wrong?

MarvFPV, 2023-04-18:
I installed everything a good... 3 weeks ago. but yea, i'll retry. With this chip i'm still dabbling around in the blinky example so the platform is set further at the top 😅

MarvFPV, 2023-04-18:
Update, same issue after reinstalling (and removing my "custom" .cfg file)

maxgerhardt, 2023-04-18:
I'm having the feeling the wrong OpenOCD version is used. I need to check some things.

maxgerhardt, 2023-04-18:
The latest published version does indeed not have a gd32e50x.cfg. This likely means I screwed up and haven't yet published the latest version into the PIO registry. I need to double check this on my computer later.

MarvFPV, 2023-04-18:
almost, after also not deleting the .piopm file^^ 

I can flash code now but debugging using cmsis-dap still does not work. i get a `unable to find a matching CMSIS-DAP device` error.

MarvFPV, 2023-04-18:
Interesting, after trying a second time it works. flashing also doesn't seem to always work... probably wiring issues... i'll play around a bit more and report if i find a pattern.

maxgerhardt, 2023-04-18:
If `upload_protocol = cmsis-dap` does not work consistently, does `upload_protocol = gdlinkcli` work? This uses the original GigaDevice upload tool. (But no debugging possible with that tool)

MarvFPV, 2023-04-20:
@maxgerhardt last question(ish), do you have an estimate when the changes will be added to the release version?

maxgerhardt, 2023-04-20:
I'll be publishing the up-to-date packages today, but that just sends them in "for review", after which by experience it'll take 1-2 weeks to appear in the PIO registry.

josedelinux, 2024-08-15:
Please update OpenOCD to 0.12.0, current version 0.11.0 failed to upload on my cheap Chinese dap debugger

maxgerhardt, 2024-08-16:
The platform mostly uses STM32 scripts anyways for GD32. Can you manually overwrite the files in `C:\Users\<user>\.platformio\packages\tool-openocd-gd32` with your 0.12.0 version?

josedelinux, 2024-08-16:
> manually overwrite the files in `C:\Users\<user>\.platformio\packages\tool-openocd-gd32` with your 0.12.0 version?
Yeah, it works when I overwrite the `bin` directory with binary copy from `tool-openocd`
thanks

closed 2025-08-15 by maxgerhardt