CommunityGD32Cores/platform-gd32 #47: GD32F303ZET6 is not support Arduino?

TimeKeepper, 2023-08-12:
Hello! I would like to develop GD32F303ZET6 using Arduino. Unfortunately, in PlatformIO, I noticed that only the standard peripheral library seems to be supported. I was wondering if there is a way to use Arduino for development with this MCU, or if you have any plans to update the Arduino framework for this MCU in the future.Thank you.

TimeKeepper, 2023-08-12:
In the board, there is a 'GD32303E-EVAL' that supports Arduino, but this version seems to have many missing pin definitions.

maxgerhardt, 2023-08-15:
Yes, only the Arduino header pins of that board are exposed, but not the full F303ZE pins. F30x boards have yet to be added to the auato-generate script with full support (https://github.com/CommunityGD32Cores/platform-gd32/blob/main/misc/scripts/gd32_genpinmap/). I am planing to do this soon.

TimeKeepper, 2023-08-16:
@maxgerhardt Received, thank you for your response and contribution.:)

closed 2025-08-15 by maxgerhardt