CommunityGD32Cores/platform-gd32 #48: Why are we using F407 flag for F405 CPU?

anisyanka, 2023-08-14:
Hello, I am not an expert, but it looks like mistake:
https://github.com/CommunityGD32Cores/platform-gd32/blob/main/boards/genericGD32F405RG.json#L5
 ```
"extra_flags": "-DGD32F4 -DGD32F407 -DGD32F4xx",
```

Might be we should use DGD32F**405**?

When I use board = genericGD32F405RG, I face with plenty of redeclarations.
For instance:
```
In file included from lib/FreeRTOS/src/cmsis_os2/os_systick.c:28:
/Users/anisyanka/.platformio/packages/framework-spl-gd32/gd32/cmsis/variants/gd32f4xx/gd32f4xx.h:248:5: error: redeclaration of enumerator 'SDIO_IRQn'
  248 |     SDIO_IRQn                    = 49,     /*!< SDIO interrupt                                           */
      |     ^~~~~~~~~
/Users/anisyanka/.platformio/packages/framework-spl-gd32/gd32/cmsis/variants/gd32f4xx/gd32f4xx.h:214:5: note: previous definition of 'SDIO_IRQn' was here
  214 |     SDIO_IRQn                    = 49,     /*!< SDIO interrupt                                           */
      |     ^~~~~~~~~
/Users/anisyanka/.platformio/packages/framework-spl-gd32/gd32/cmsis/variants/gd32f4xx/gd32f4xx.h:249:5: error: redeclaration of enumerator 'TIMER4_IRQn'
  249 |     TIMER4_IRQn                  = 50,     /*!< TIMER4 interrupt                                         */
      |     ^~~~~~~~~~~
...
```

maxgerhardt, 2023-08-14:
[image]

Unbelievable! The CSV data I obtained from **GIGADEVICE** is wrong?!

maxgerhardt, 2023-08-14:
Fixed in https://github.com/CommunityGD32Cores/platform-gd32/commit/8df39688433abebb00a93c17c53cd2372cd482b6

Update with [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) -> `pio pkg update -g -p gd32`

closed 2023-08-14 by maxgerhardt

anisyanka, 2023-08-14:
It works, thanks!